suaveolent/ha-hoymiles-wifi #87: Why has the power limit functionality been deactivated for many inverters?

justinh998, 2025-01-06:
The changelog of the 0.3.3 version states: 
```
Enabled power limit functionality only for the HMS-W (DTUBI) series.
```

But why has this been disabled for all other inverter types? It worked there too.

suaveolent, 2025-01-06:
This was mainly a precautionary measure. 
I don’t have a dedicated DTU, so I don’t know what happens if this is used for multiple inverters. 

Can you test that?

elmodem, 2025-01-19:
> This was mainly a precautionary measure. I don’t have a dedicated DTU, so I don’t know what happens if this is used for multiple inverters.
> 
> Can you test that?

I'm also interested in getting this functionality back since the local power network operators are looking more and more at the electricity feed-in from balcony power plant owners (because of power grid overload one afternoon).

Can you assist one how to test.
Running a WLite-S DTU connected to a HMS-1800-4T.

suaveolent, 2025-01-20:
Basically I would need someone with a WLite and multiple inverters. Currently I am using the command from the HMS-W-Series. However, I don't know what happens if I send this command to a DTU with multiple inverters. E.g. does this limit the power of all inverters?

elmodem, 2025-01-20:
Got it. 
Unfortunately, I currently only have one inverter on the WLite. However, I also use a Hoymiles Inverter W and I noticed something interesting:
- With Hoymiles W models, the inverter and DTU have the same device ID.
- With a WLite connected to a Hoymiles Inverter there are different device IDs

suaveolent, 2025-01-20:
> Got it. Unfortunately, I currently only have one inverter on the WLite. However, I also use a Hoymiles Inverter W and I noticed something interesting:
> 
> * With Hoymiles W models, the inverter and DTU have the same device ID.
> * With a WLite connected to a Hoymiles Inverter there are different device IDs for the Inverter and the DTU

Yes that is correct. The W series has an integrated DTU (there is only one digit changing for the DTU and the inverter).

elmodem, 2025-01-20:
I cannot find any power limit settings at the DTU or inverter level. The S-Miles Cloud only seems to process the command at the level of the entire plant. Since an entire plant always consists of the DTU and the assigned inverter, this consideration should be irrelevant.
In principle, it is possible to connect several inverters with the WLite-S. However, the limitation is a total of 4 PV modules overall.
- max. 4 inverter with 1 PV module each
- 2 inverter with 1-2 PV module each
- Most likely: 1 Inverter with 2-4 modules

justinh998, 2025-01-20:
> I cannot find any power limit settings at the DTU or inverter level. The S-Miles Cloud only seems to process the command at the level of the entire plant. Since an entire plant always consists of the DTU and the assigned inverter, this consideration should be irrelevant.
> In principle, it is possible to connect several inverters with the WLite-S. However, the limitation is a total of 4 PV modules overall.
> - max. 4 inverter with 1 PV module each
> - 2 inverter with 1-2 PV module each
> - Most likely: 1 Inverter with 2-4 modules
> 
> This only applies to the WLite-S model!
> 

But this also applies to the w series, so if I have 4 w series inverters with 1 module each and they are all in one plant, it is the same situation.

elmodem, 2025-01-20:
W series comes with an integrated DTU. 1 inverter = 1 DTU.

suaveolent, 2025-01-20:
> I cannot find any power limit settings at the DTU or inverter level. The S-Miles Cloud only seems to process the command at the level of the entire plant. Since an entire plant always consists of [...]

Alright, got it. 

In the end this, however, would mean that the power limit is set on DTU level only. 
So if I have, e.g. two DTU's (e.g. one DTU-Wlite (with multiple inverters) and one W-Series inverter) I would need to set the power limit for both DTU's separately.

elmodem, 2025-01-21:
Thanks. So far nothing changed, even after updating and reinstalling via Hacs. 
No entity to control the power limitation for the DTU or the Inverter. Did I missed aomething?

suaveolent, 2025-01-22:
> Thanks. So far nothing changed, even after updating and reinstalling 0.3.3 via Hacs. No entity to control the power limitation for the DTU or the Inverter. Did I missed something?
> 
> I went back to 0.3.2 main and now it's showing up.

There is no new version yet. The next version with these fixes will be `0.3.4`. If you want to test them, you need to manually update to the main branch.

To manually update to the `main` branch you need to go to "Developer Tools" - "Actions":
- Actions: "Update: install update"
- Targets: "Hoymiles update"
- Version: "main"

Afterwards there should be a restart displayed under "Settings" (Note: The version should be 606d5a7)

suaveolent, 2025-01-27:
Should be fixed in latest release

closed 2025-01-27 by suaveolent