suaveolent/ha-hoymiles-wifi #88: Support of other Hoymiles inverters like HM series and the new HAT / HYT / HAS / HYS battery inverter

hangingloose, 2025-01-06:
Hello and thanks for the integation!

In my system I have HMT / HM and HYS inverter from Hoymiles.

Is it possible to implement all of them with this integration? 
If not yet, is there a plan to add them into the integration?

Thanks

hangingloose, 2025-01-06:
Thanks for the quick answer. 

For my DTU lite and 4 HMT the integration ist working fine.
My DTU Pro (without S) for my HM is in the same network and has a problem. 
I get an error, "connection not possible" The host address is fine!

the same for my DTS of my HYS. 
For the toolkit I have to chose the network of the DTU. 
for the Host address the device has to be in my local network.

suaveolent, 2025-01-07:
If your DTU Pro is connected via Ethernet then this issue might be related to:
https://github.com/suaveolent/ha-hoymiles-wifi/issues/73

Afaik there might be an update necessary, otherwise connection will only work for WiFi connections.

As far as hostname goes: This is most likely an issue with your local network since it merely uses the underlying library to get the IP. Are you sure you are using the correct hostname?

Unfortunately I do not have a HYS inverter. Does it have an integrated DTU and does it expose a wifi hotspot?

hangingloose, 2025-01-07:
Hello Suaveolent, 

my DTU Pro is as the lite connected via WLAN. 
[image]
The Pro has the IP 173
DTU | Online | Gen3 | DTU-Pro |   | H09.02.12 | V00.04.03
The Lite has the IP 215 
DTU | Online | Gen3 | DTU-Lite-S |   | H06.02.01 | V00.04.00

The name of the DTU from the HYS ist DTS. The DTS has also his own WLAN and is connected to my WLAN with the IP 101

suaveolent, 2025-01-09:
Trying to split up the conversation:

`HYS`:
Alright, thanks for the clarification. Just to clarify once more: If you connect to your `HYS`, you can view live data view the toolkit in the app right? I am not talking about the e.g. 15 minutes data provided via the hoymiles cloud.

`Connection issues`
The IP address of the home assistant instance should not matter.
To better debug these issues could you try to issue the `get-real-data-new` command from the https://github.com/suaveolent/hoymiles-wifi to your DTU Pro? e.g.

```
hoymiles-wifi --host 192.168.2.173  get-real-data-new
```

hangingloose, 2025-01-09:
Hello,
HYS: yes I am able to get a connection to the DTU with the toolkit. I only have to choose the right network. 

Connection:
I try to help you debug, but I don’t know how... 
Where should I copy 

hoymiles-wifi --host 192.168.2.173  get-real-data-new

CMD? HA-Terminal? (both are not working) 
Sorry for this question, but I didn´t found an information about this command.

suaveolent, 2025-01-09:
HYS: 

great. However, what I am missing from the screenshot is whether there are any live data you Can retrieve from it. E.g current power or whatever. 

connection: 
That is great news. A different computer should not make a difference since it’s always the home assistant instance making the request. Maybe the firmware update and some reboot did the trick.

hangingloose, 2025-01-10:
If I connect the DTU with the toolkit I get this pictures:
 
[image]
[image]
[image]
There are some more information about the battery

Also interesting are:
[image]

suaveolent, 2025-01-10:
Alright this is great thanks. This means that there should be a way to integrate these inverters. I will have a look at the code to see how this is done.

hangingloose, 2025-01-10:
do you need more information?

hangingloose, 2025-01-10:
Then I need a short support. Where should i type in the command? 
Sorry but i have no idea...
I tried CMD and Terminal @HA

suaveolent, 2025-01-11:
Here is what you need to do:

1. Ensure you python installed (most likely your terminal at home assistant, since you seem to be running Windows)
2. Enter the following command `pip install hoymiles-wifi`
3. If your HYS is connected to your home wifi, enter the following command: `hoymiles-wifi --host <IP_OF_YOUR_HYS> get-real-data-new`
4. Enter also this command `hoymiles-wifi --host <IP_OF_YOUR_HYS> app-information-data`

hangingloose, 2025-01-11:
Thanks for this information. It fits!
But I am sorry, 
for both command I get the response “No response or unable to retrieve response for …. “
The DTU of my HYS hat a password protection for the own WLAN hotspot. 
This could not be the problem, or`?

suaveolent, 2025-01-27:
> Thanks for this information. It fits! But I am sorry, for both command I get the response “No response or unable to retrieve response for …. “ The DTU of my HYS hat a password protection for the own WLAN hotspot. This could not be the problem, or`?

No I don't think that the password of the HYS should play a role here. It just might be the command is not the correct one for the HYS. 
I am currently trying to decompile the new version of the app, however there seems to be an issue with the underlying toolset. Until then, there is not much I can do unfortunately.

hangingloose, 2025-02-05:
Can I support you with additional information?

suaveolent, 2025-02-10:
I am waiting for https://github.com/iBotPeaches/Apktool/issues/3776 to be resolved, s.t. I can have a look at the newest apk from hoymiles.

Since the existing requests do not work, I expect that there was some additional support added.

magyarl89, 2025-02-15:
I'm very excited about this topic too, because I have HYT-HV inverter with battery and DTS-WIFI-G1 Dongle was included.
This would be a great solution for many of us, because we just have had investment support from the hungarian goverment for hybrid systems, so a few hundred HYT-HV was installed.
I have the same experience as @hangingloose: connecting to the hotspot of the dongle, I can see the live data with the installer app's toolkit. However I have no luck to use this integration, even if I connect my HA instance directly to the dongle hotspot.

suaveolent, 2025-02-18:
Just letting you know that I made some progress with decompiling the app. I will now have a look whethere there is anything useful for communicating with the battery.

hangingloose, 2025-02-18:
If you want us to test something, let us know ;-)

suaveolent, 2025-02-18:
> If you want us to test something, let us know ;-)

I am currently implementing what I think is required to communicate with the battery :) . Once that is done, I am more than happy if you could test it.

suaveolent, 2025-02-18:
> If you want us to test something, let us know ;-)

if you have access to Wireshark an  can capture the requests of course, this would also help, in case I missed something.

hangingloose, 2025-02-18:
Sorry, I installed Wireshark on my laptop, but that doesn't help me because the communication between the cell phone and HYS does not take place via the router, it goes via the WLAN from the HYS.
I cannot install Wireshark on my company mobile phone.
I don't have the toolkit on my laptop ;-(

suaveolent, 2025-02-18:
> Sorry, I installed Wireshark on my laptop, but that doesn't help me because the communication between the cell phone and HYS does not take place via the router, it goes via the WLAN from the HYS. I cannot install Wireshark on my company mobile phone. I don't have the toolkit on my laptop ;-(

I just pushed a test version to the `bms` branch of `hoymiles-wifi`. Are you familiar with python and can you test the newly added `get-energy-storage-registry` command?

https://github.com/suaveolent/hoymiles-wifi/tree/bms

hangingloose, 2025-02-18:
not really, so far I've been following instructions ;-)
Do I have to replace the data or install?
otherwise small instructions possible?

hangingloose, 2025-02-18:
Problem on step 7:

➜  hoymiles-wifi-bms pip3 install dist/hoymiles_wifi-0.3.5-py3-none-any.whl --force-reinstall

-WARNING: Requirement 'dist/hoymiles_wifi-0.3.5-py3-none-any.whl' looks like a filename, but the file does not exist
-Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
-Processing ./dist/hoymiles_wifi-0.3.5-py3-none-any.whl
-ERROR: Could not install packages due to an OSError: [Errno 2] No such file or directory: '/homeassistant/hoymiles-wifi-bms/dist/hoymiles_wifi-0.3.4-py3-none-any.whl'

suaveolent, 2025-02-18:
> Problem on step 7:
> 
> ➜ hoymiles-wifi-bms pip3 install dist/hoymiles_wifi-0.3.4-py3-none-any.whl --force-reinstall
> 
> -WARNING: Requirement 'dist/hoymiles_wifi-0.3.4-py3-none-any.whl' looks like a filename, but the file does not exist -Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/ -Processing ./dist/hoymiles_wifi-0.3.4-py3-none-any.whl -ERROR: Could not install packages due to an OSError: [Errno 2] No such file or directory: '/homeassistant/hoymiles-wifi-bms/dist/hoymiles_wifi-0.3.4-py3-none-any.whl'

Also forgot to enable the `venv`:

`source .venv/bin/activate`

suaveolent, 2025-02-18:
> Everything worked fine up to step 10! Problem, my HomeAssistant PC on which I installed and entered it has no WiFi and is not within range of the HYS... ;-( I have to connect the PC to the internal WLAN of the HYS, right?

Or alternatively, if your HYS is connected to your home wifi, you could also try this IP address.

The reason I was asking to directly connect to the HYS is just to rule out any network issues.

hangingloose, 2025-02-18:
(.venv) ➜  hoymiles-wifi-bms hoymiles-wifi --host 192.168.2.101 get-energy-storage-registry
usage: hoymiles-wifi [-h] --host HOST [--local_addr LOCAL_ADDR] [--as-json] [--disable-interactive] [--power-limit POWER_LIMIT]
                     {get-real-data-new,get-real-data,get-config,network-info,app-information-data,app-get-hist-power,set-power-limit,set-wifi,firmware-update,restart-dtu,turn-on-inverter,turn-off-inverter,get-information-data,get-version-info,heartbeat,identify-dtu,identify-inverters,identify-meters,get-alarm-list,enable-performance-data-mode}
hoymiles-wifi: error: argument command: invalid choice: 'get-energy-storage-registry' (choose from get-real-data-new, get-real-data, get-config, network-info, app-information-data, app-get-hist-power, set-power-limit, set-wifi, firmware-update, restart-dtu, turn-on-inverter, turn-off-inverter, get-information-data, get-version-info, heartbeat, identify-dtu, identify-inverters, identify-meters, get-alarm-list, enable-performance-data-mode)

??? do I have an old version, or do I need a restart?

suaveolent, 2025-02-18:
> (.venv) ➜ hoymiles-wifi-bms hoymiles-wifi --host 192.168.2.101 get-energy-storage-registry usage: hoymiles-wifi [-h] --host HOST [--local_addr LOCAL_ADDR] [--as-json] [--disable-interactive] [...]

You need to make sure that you enable the `venv` before building and installing the library. Otherwise you have installed the library system wide. (Check the instructions again, I have previously updated it to show when you need to enable the venv)

hangingloose, 2025-02-18:
crazy!
As soon as you do it right, it works ;-)

But is this the result you are waiting for?

(.venv) ➜  hoymiles-wifi-bms hoymiles-wifi --host 192.168.2.101 get-energy-storage-registry 
Traceback (most recent call last):
  File "/homeassistant/hoymiles-wifi-bms/.venv/bin/hoymiles-wifi", line 8, in <module>
    sys.exit(run_main())
             ^^^^^^^^^^
  File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/__main__.py", line 470, in run_main
    asyncio.run(main())
  File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 686, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/__main__.py", line 439, in main
    response = await command_func(dtu)
               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/__main__.py", line 333, in async_get_energy_storage_registry
    return await dtu.async_get_energy_storage_registry()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/dtu.py", line 391, in async_get_energy_storage_registry
    request.time_ymd_hms = (
    ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ESRegResDTO' object has no attribute 'time_ymd_hms'

suaveolent, 2025-02-18:
> crazy! As soon as you do it right, it works ;-)
> 
> But is this the result you are waiting for?
> 
> (.venv) ➜ hoymiles-wifi-bms hoymiles-wifi --host 192.168.2.101 get-energy-storage-registry Traceback (most recent call last): File "/homeassistant/hoymiles-wifi-bms/.venv/bin/hoymiles-wifi", line 8, in sys.exit(run_main()) ^^^^^^^^^^ File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/**main**.py", line 470, in run_main asyncio.run(main()) File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run return runner.run(main) ^^^^^^^^^^^^^^^^ File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run return self._loop.run_until_complete(task) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/lib/python3.12/asyncio/base_events.py", line 686, in run_until_complete return future.result() ^^^^^^^^^^^^^^^ File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/**main**.py", line 439, in main response = await command_func(dtu) ^^^^^^^^^^^^^^^^^^^^^^^ File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/**main**.py", line 333, in async_get_energy_storage_registry return await dtu.async_get_energy_storage_registry() ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/dtu.py", line 391, in async_get_energy_storage_registry request.time_ymd_hms = ( ^^^^^^^^^^^^^^^^^^^^ AttributeError: 'ESRegResDTO' object has no attribute 'time_ymd_hms'

:) 

Right, I made some fixes but forgot to push them.
Unfortunately, you need to repeat all steps again to have the new fixes.

hangingloose, 2025-02-18:
The files at 
https://github.com/suaveolent/hoymiles-wifi/tree/bms

are still 3h old.
should I wait a little bit?

suaveolent, 2025-02-18:
> The files at https://github.com/suaveolent/hoymiles-wifi/tree/bms
> 
> are still 3h old. should I wait a little bit?

No it's fine. The fix was made 3h ago, but I forgot to push it. Latest commit it "[fix naming](https://github.com/suaveolent/hoymiles-wifi/commit/5e82ab579ec66284ebf5cfa09fe725098f9ba2f6)"

hangingloose, 2025-02-18:
(.venv) ➜  hoymiles-wifi-bms hoymiles-wifi --host 192.168.2.101 get-energy-storage-registry          
No response or unable to retrieve response for get-energy-storage-registry

suaveolent, 2025-02-18:
> (.venv) ➜ hoymiles-wifi-bms hoymiles-wifi --host 192.168.2.101 get-energy-storage-registry No response or unable to retrieve response for get-energy-storage-registry

Hm that's unfortunate. I'll keep looking for relevant parts in the code.

Just to make sure: You are certain that `192.168.2.101` is the right IP? Could you maybe also try to connect directly to the DTU to see if the command works?

magyarl89, 2025-02-19:
I managed to test it, but I got the "No response" message too.
I tested it on my Windows laptop and I run it with direct connection and through the my router. In both cases ping were successful.
I had a little hassle to figure out how to use the given steps in Windows CMD, so here you can check, if I did it right:

`Microsoft Windows [Version 10.0.22631.4751]
(c) Microsoft Corporation. Minden jog fenntartva.

C:\Users\mly2bp>cd c:\temp\hoymiles-wifi-bms\

c:\temp\hoymiles-wifi-bms>python -m venv .venv

c:\temp\hoymiles-wifi-bms>cd C:\temp\hoymiles-wifi-bms\.venv\Scripts\

C:\temp\hoymiles-wifi-bms\.venv\Scripts>activate.bat

(.venv) C:\temp\hoymiles-wifi-bms\.venv\Scripts>cd c:\temp\hoymiles-wifi-bms\

(.venv) c:\temp\hoymiles-wifi-bms>pip3 install build
Collecting build
  Downloading build-1.2.2.post1-py3-none-any.whl.metadata (6.5 kB)
Collecting packaging>=19.1 (from build)
  Downloading packaging-24.2-py3-none-any.whl.metadata (3.2 kB)
Collecting pyproject_hooks (from build)
  Downloading pyproject_hooks-1.2.0-py3-none-any.whl.metadata (1.3 kB)
Collecting colorama (from build)
  Downloading colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Downloading build-1.2.2.post1-py3-none-any.whl (22 kB)
Downloading packaging-24.2-py3-none-any.whl (65 kB)
Downloading colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Downloading pyproject_hooks-1.2.0-py3-none-any.whl (10 kB)
Installing collected packages: pyproject_hooks, packaging, colorama, build
Successfully installed build-1.2.2.post1 colorama-0.4.6 packaging-24.2 pyproject_hooks-1.2.0

[notice] A new release of pip is available: 24.3.1 -> 25.0.1
[notice] To update, run: python.exe -m pip install --upgrade pip

(.venv) c:\temp\hoymiles-wifi-bms>python3 -m build --wheel
A Python nem talßlhat≤. A Microsoft Store-b≤l val≤ telepφtΘshez futtassa a parancsot argumentumok nΘlkⁿl, vagy a Beßllφtßsok > Alkalmazßs-vΘgrehajtßsi aliasok kezelΘse lehet
(.venv) c:\temp\hoymiles-wifi-bms>python -m build --wheel
* Creating isolated environment: venv+pip...
* Installing packages in isolated environment:
  - setuptools >= 40.8.0
* Getting build dependencies for wheel...
running egg_info
creating hoymiles_wifi.egg-info
writing hoymiles_wifi.egg-info\PKG-INFO
writing dependency_links to hoymiles_wifi.egg-info\dependency_links.txt
writing entry points to hoymiles_wifi.egg-info\entry_points.txt
writing requirements to hoymiles_wifi.egg-info\requires.txt
writing top-level names to hoymiles_wifi.egg-info\top_level.txt
writing manifest file 'hoymiles_wifi.egg-info\SOURCES.txt'
reading manifest file 'hoymiles_wifi.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE'
writing manifest file 'hoymiles_wifi.egg-info\SOURCES.txt'
* Building wheel...
running bdist_wheel
running build
running build_py
creating build\lib\hoymiles_wifi
copying hoymiles_wifi\[const.py](http://const.py/) -> build\lib\hoymiles_wifi
copying hoymiles_wifi\[dtu.py](http://dtu.py/) -> build\lib\hoymiles_wifi
copying hoymiles_wifi\[hoymiles.py](http://hoymiles.py/) -> build\lib\hoymiles_wifi
copying hoymiles_wifi\[utils.py](http://utils.py/) -> build\lib\hoymiles_wifi
copying hoymiles_wifi\[__init__.py](http://__init__.py/) -> build\lib\hoymiles_wifi
copying hoymiles_wifi\[__main__.py](http://__main__.py/) -> build\lib\hoymiles_wifi
creating build\lib\hoymiles_wifi\protobuf
copying hoymiles_wifi\protobuf\[AlarmData_pb2.py](http://alarmdata_pb2.py/) -> build\lib\hoymiles_wifi\protobuf
copying hoymiles_wifi\protobuf\[AppGetHistED_pb2.py](http://appgethisted_pb2.py/) -> build\lib\hoymiles_wifi\protobuf
copying hoymiles_wifi\protobuf\[AppGetHistPower_pb2.py](http://appgethistpower_pb2.py/) -> build\lib\hoymiles_wifi\protobuf
[... 100 lines cut ...]
Processing .\dist\hoymiles_wifi-0.3.4-py3-none-any.whl
Collecting protobuf (from hoymiles-wifi==0.3.4)
  Downloading protobuf-5.29.3-cp310-abi3-win_amd64.whl.metadata (592 bytes)
Collecting crcmod (from hoymiles-wifi==0.3.4)
  Downloading crcmod-1.7.tar.gz (89 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Downloading protobuf-5.29.3-cp310-abi3-win_amd64.whl (434 kB)
Building wheels for collected packages: crcmod
  Building wheel for crcmod (pyproject.toml) ... done
  Created wheel for crcmod: filename=crcmod-1.7-py3-none-any.whl size=18916 sha256=34377e789e81a28a769bad684baa6f767558aaed6d951a8076673fa453ae971c
  Stored in directory: c:\users\mly2bp\appdata\local\pip\cache\wheels\c5\b4\7d\51d3485a8022501c7f7dba78b6c945088ff93d40142c9a96f4
Successfully built crcmod
Installing collected packages: crcmod, protobuf, hoymiles-wifi
Successfully installed crcmod-1.7 hoymiles-wifi-0.3.4 protobuf-5.29.3

[notice] A new release of pip is available: 24.3.1 -> 25.0.1
[notice] To update, run: python.exe -m pip install --upgrade pip

(.venv) c:\temp\hoymiles-wifi-bms>hoymiles-wifi --host 10.10.100.254 get-energy-storage-registry
No response or unable to retrieve response for get-energy-storage-registry

(.venv) c:\temp\hoymiles-wifi-bms>hoymiles-wifi --host 192.168.1.183 get-energy-storage-registry
No response or unable to retrieve response for get-energy-storage-registry`

magyarl89, 2025-02-19:
Hack yeah, I managed to sniff the packets send and received by the Installer app. PCAPdroid is a cool app to capture. (even without root!)
@suaveolent where should I send you my packets? (as it may contain some sensitive data)

suaveolent, 2025-02-19:
> Hack yeah, I managed to sniff the packets send and received by the Installer app. PCAPdroid is a cool app to capture. (even without root!) [@suaveolent](https://github.com/suaveolent) where should I send you my packets? (as it may contain some sensitive data)

That’s great news! Thanks! 

You can send it to: 

<my username> at proton . me

suaveolent, 2025-02-19:
> Hack yeah, I managed to sniff the packets send and received by the Installer app. PCAPdroid is a cool app to capture. (even without root!) [@suaveolent](https://github.com/suaveolent) where should I send you my packets? (as it may contain some sensitive data)

Thanks for the dump. That really helps a lot!
Interestingly it seems to send exactly the same data as the lib now does (beginning with `c3 02`). Which confirms at least my implementation.
I now need to investigate whether some other packages are required before that.

suaveolent, 2025-02-25:
@magyarl89 @hangingloose 

I just pushed a new version to to the `bms` branch adding two new commands:

`get-gateway-info` and `get-gateway-network-info`

Can you check wethere either of these commands return anything useful?

hangingloose, 2025-02-25:
> Alright here is what you need to do:
> 
> 1. Go to https://github.com/suaveolent/hoymiles-wifi/tree/bms
> 2. Click "Code" - "Download ZIP"
> 3. Unzip the downloaded file
> 4. Open a Command Line (CMD), navigate to the unpacked directory
> 5. Type: `python3 -m venv .venv`
> 6. Type: `source .venv/bin/activate`
> 7. Type: `pip3 install build`
> 8. Type: `python3 -m build --wheel`
> 9. Type: `pip3 install dist/hoymiles_wifi-0.3.4-py3-none-any.whl --force-reinstall`
> 10. Connect directly to the WiFi of the Battery
> 11. Confirm that the DTU has the IP-Address `10.10.100.254`
> 12. Type: `hoymiles-wifi --host 10.10.100.254 get-energy-storage-registry` (or whathever IP address your DTU has)

looks like I have a problem.

(.venv) ➜  /config python3 -m venv .venv    
(.venv) ➜  /config source .venv/bin/activate
(.venv) ➜  /config pip3 install build       
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Requirement already satisfied: build in ./.venv/lib/python3.12/site-packages (1.2.2.post1)
Requirement already satisfied: packaging>=19.1 in ./.venv/lib/python3.12/site-packages (from build) (24.2)
Requirement already satisfied: pyproject_hooks in ./.venv/lib/python3.12/site-packages (from build) (1.2.0)
(.venv) ➜  /config python3 -m build --wheel 
* Creating isolated environment: venv+pip...
**ERROR Source** /homeassistant does not appear to be a Python project: no pyproject.toml or setup.py

hangingloose, 2025-02-25:
Successfully built hoymiles_wifi-0.3.4-py3-none-any.whl
(.venv) ➜  hoymiles-wifi-bms **pip3 install dist/hoymiles_wifi-0.3.4-py3-none-any.whl --force-reinstall**
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Processing ./dist/hoymiles_wifi-0.3.4-py3-none-any.whl
Collecting protobuf (from hoymiles-wifi==0.3.4)
  Downloading protobuf-5.29.3-py3-none-any.whl.metadata (592 bytes)
Collecting crcmod (from hoymiles-wifi==0.3.4)
  Downloading https://wheels.home-assistant.io/musllinux-index/crcmod-1.7-cp312-cp312-musllinux_1_2_x86_64.whl.metadata (5.2 kB)
Downloading https://wheels.home-assistant.io/musllinux-index/crcmod-1.7-cp312-cp312-musllinux_1_2_x86_64.whl (23 kB)
Downloading protobuf-5.29.3-py3-none-any.whl (172 kB)
Installing collected packages: crcmod, protobuf, hoymiles-wifi
Successfully installed crcmod-1.7 hoymiles-wifi-0.3.4 protobuf-5.29.3
(.venv) ➜  hoymiles-wifi-bms **hoymiles-wifi --host 192.168.2.101 get-energy-storage-registry**          
No response or unable to retrieve response for get-energy-storage-registry
(.venv) ➜  hoymiles-wifi-bms **hoymiles-wifi --host 192.168.2.101 get-gateway-info**                            
No response or unable to retrieve response for get-gateway-info
(.venv) ➜  hoymiles-wifi-bms **hoymiles-wifi --host 192.168.2.101 get-gateway-network-info**
No response or unable to retrieve response for get-gateway-network-info
(.venv) ➜  hoymiles-wifi-bms

suaveolent, 2025-02-25:
> Successfully built hoymiles_wifi-0.3.4-py3-none-any.whl (.venv) ➜ hoymiles-wifi-bms **pip3 install dist/hoymiles_wifi-0.3.4-py3-none-any.whl --force-reinstall** Looking in indexes: [...]

@hangingloose Thanks for the quick reply. Thats really unfortunate!

According to the dumps provided by @magyarl89 these should be the right commands.

They are send by the app in the following order:

`get-gateway-info`
`get-gateway-network-info`
`get-energy-storage-registry`

Maybe it only works when directly connected to the DTU. I will also check if there is anything else that has changed.

suaveolent, 2025-02-25:
@hangingloose  @magyarl89 I just discovered that there seems to be a bug when trying to parse the response. Once I have fixed it I will let you know.

hangingloose, 2025-02-25:
bull's-eye! 
can you give me your email, then I will send you the response

suaveolent, 2025-02-25:
@hangingloose great thanks! Next step will be to retrieve the live data!

suaveolent, 2025-02-26:
@hangingloose @magyarl89 I have added a `get-energy-storage-data`  request. Can you check if it works?

hangingloose, 2025-02-26:
(.venv) ➜  hoymiles-wifi-bms hoymiles-wifi --host 192.168.2.101 get-energy-storage-data                
Traceback (most recent call last):
  File "/homeassistant/hoymiles-wifi-bms/.venv/bin/hoymiles-wifi", line 8, in <module>
    sys.exit(run_main())
             ^^^^^^^^^^
  File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/__main__.py", line 500, in run_main
    asyncio.run(main())
  File "/usr/lib/python3.12/asyncio/runners.py", line 195, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 691, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/__main__.py", line 469, in main
    response = await command_func(dtu)
               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/hoymiles-wifi-bms/.venv/lib/python3.12/site-packages/hoymiles_wifi/__main__.py", line 343, in async_get_energy_storage_data
    return await dtu.async_get_energy_storage_data(registry.serial_number)
                                                   ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ESRegReqDTO' object has no attribute 'serial_number'

magyarl89, 2025-02-26:
I have the same error response for both router and direct wifi connection using Windows laptop:

(.venv) c:\temp\hoymiles-wifi-bms>hoymiles-wifi --host 10.10.100.254 get-energy-storage-registry
Get-energy-storage-registry Response:
type: 3
meter {
  sn: "1234"
}
inv {
  serial_number: 1234
  sw_m_ver: "V01.02.29"
  sw_s_ver: "V01.04.00"
  sw_sys_ver: "V02.01.03"
  bms_cap: 100
  pv_num: 2
  addr: 1
  model_name: "HYT-5.0HV-EUG1"
  bms_type: 1
}
time: 1740604530
offset: 3600
dfs: 131088

(.venv) c:\temp\hoymiles-wifi-bms>hoymiles-wifi --host 10.10.100.254 get-gateway-info
Get-gateway-info Response:
serial_number: 1234
time: 1740604507
ap: 1
mgwinfo {
  mod: 1
  sw: 4105
  hw: 45313
  c_mod: 1
  wifi_v: "2.3.0.0-dev"
  rid: 21
}
mdevinfo {
  serial_number: 1234
  type: 6
  addr: 1
}

(.venv) c:\temp\hoymiles-wifi-bms>hoymiles-wifi --host 10.10.100.254 get-gateway-network-info
Get-gateway-network-info Response:
serial_number: 1234
time: 1740604519
net_set_mod: 1
net_set_state: 18
net_mod: 1
net_time: 1740604516
csq: 100
net_state: 1
ap_set_tate: 2

(.venv) c:\temp\hoymiles-wifi-bms>hoymiles-wifi --host 10.10.100.254 get-energy-storage-data
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "c:\temp\hoymiles-wifi-bms\.venv\Scripts\hoymiles-wifi.exe\[__main__.py](http://__main__.py/)", line 7, in <module>
    sys.exit(run_main())
             ~~~~~~~~^^
  File "c:\temp\hoymiles-wifi-bms\.venv\Lib\site-packages\hoymiles_wifi\[__main__.py](http://__main__.py/)", line 500, in run_main
    asyncio.run(main())
    ~~~~~~~~~~~^^^^^^^^
  File "C:\Users\mly2bp\AppData\Local\Programs\Python\Python313\Lib\asyncio\[runners.py](http://runners.py/)", line 195, inrun
    return runner.run(main)
           ~~~~~~~~~~^^^^^^
  File "C:\Users\mly2bp\AppData\Local\Programs\Python\Python313\Lib\asyncio\[runners.py](http://runners.py/)", line 118, inrun
    return self._loop.run_until_complete(task)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "C:\Users\mly2bp\AppData\Local\Programs\Python\Python313\Lib\asyncio\[base_events.py](http://base_events.py/)", line 725, in run_until_complete
    return future.result()
           ~~~~~~~~~~~~~^^
  File "c:\temp\hoymiles-wifi-bms\.venv\Lib\site-packages\hoymiles_wifi\[__main__.py](http://__main__.py/)", line 469, in main
    response = await command_func(dtu)
               ^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\temp\hoymiles-wifi-bms\.venv\Lib\site-packages\hoymiles_wifi\[__main__.py](http://__main__.py/)", line 343, in async_get_energy_storage_data
    return await dtu.async_get_energy_storage_data(registry.serial_number)
                                                   ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'serial_number'

suaveolent, 2025-02-27:
@magyarl89 @hangingloose Right. That's what happens if you quickly try to implement something without looking at all the data. :) 

I have just pushed a new version which should hopefully correctly implement it.

suaveolent, 2025-02-27:
@magyarl89 thanks. Occured due to renaming. Could you retry?

suaveolent, 2025-02-27:
@magyarl89 you are absolutetly right. Sorry, I forgot to compile the protobuf files.  Should be fixed now.

magyarl89, 2025-02-27:
Are you sure? Same error 😅

suaveolent, 2025-02-27:
> sorry, not same: File "c:\temp\hoymiles-wifi-bms.venv\Lib\site-packages\hoymiles_wifi\dtu.py", line 458, in async_get_energy_storage_data request.time_ymd_hms = ( ^^^^^^^^^^^^^^^^^^^^ TypeError: 'bytes' object cannot be interpreted as an integer

@magyarl89  Right. Apparently hoymiles did not use the same data type for the field across their protobufs.... Let me check what they did there..

suaveolent, 2025-02-27:
@magyarl89 Can you try again?

magyarl89, 2025-02-27:
**Nice job! :D** 
What's next?

(.venv) c:\temp\hoymiles-wifi-bms>hoymiles-wifi --host 192.168.1.183 get-energy-storage-data
b'2025-02-27 15:42:41'
Get-energy-storage-data Response:
[serial_number: 1234
timestamp: 1740667398
offset: 3600
active_power: 1
production {
  energy_to_load: 45
  energy_to_battery: 35
}
consumption {
  energy_from_pv: 45
  energy_from_battery: 2
  energy_from_grid: 32
}
pv_panels {
  solar_irradiance: 1
  voltage: 5288
  current: 126
  power: 439
  energy: 79
}
pv_panels {
  solar_irradiance: 2
  voltage: 383
  current: -6
}
battery_management {
  type: 1
  state_of_charge: 45
  state_of_health: 99
  voltage: 2386
  internal_charge_mode: 250
  internal_discharge_mode: 250
  cell_voltage_high: 2592
  cell_voltage_low: 2016
  temp_high_charge: 240
  temp_low_charge: 200
  temp_high_module: 240
  temp_low_module: 200
  energy_charged: 38
  energy_discharged: 2
  voltage_charge_high: 3309
  voltage_charge_low: 3305
  voltage_module_high: 3970
  voltage_module_low: 3968
}
grid {
  param {
    status: 1
    frequency: 4998
    power_factor_deviation: -1
  }
  phases {
    phase: 1
    voltage: 2329
    current: 46
    reactive_power: -22
    power_factor: -17
    energy_frequency: 1
    energy_consumed: 12
  }
  phases {
    phase: 2
    voltage: 2314
    current: 84
    reactive_power: 45
    power_factor: -1
    energy_consumed: 13
  }
  phases {
    phase: 3
    voltage: 2345
    current: 45
    reactive_power: 71
    power_factor: 15
    energy_frequency: 1
    energy_consumed: 11
  }
}
load {
  param {
    status: 1
    frequency: 4998
  }
  phases {
    phase: 1
    voltage: 2316
    active_power: 139
    energy_consumed: 31
  }
  phases {
    phase: 2
    voltage: 2306
    active_power: 219
    energy_consumed: 39
  }
  phases {
    phase: 3
    voltage: 2352
    active_power: 81
    energy_consumed: 8
  }
}
inverter {
  param {
    status: 3
    frequency: 4997
    isolation_resistance: 4980
    leakage_current: -1
    drm_signal: 1
  }
  phases {
    phase: 1
    voltage: 2327
    current: 98
    active_power: 100
    reactive_power: -11
    dc_current: -19
    dc_voltage: -207
    eps_voltage: 28
    eps_current: -64
    eps_power: -1
  }
  phases {
    phase: 2
    voltage: 2310
    current: 126
    active_power: 212
    reactive_power: -5
    dc_current: -16
    dc_voltage: 148
    eps_voltage: 50
    eps_current: -85
    eps_power: -4
  }
  phases {
    phase: 3
    voltage: 2345
    current: 85
    active_power: 68
    reactive_power: 8
    dc_current: -13
    dc_voltage: -51
    eps_voltage: 33
    eps_current: -67
    eps_power: -2
  }
}
generator {
  param {
  }
  phases {
    phase: 1
  }
  phases {
    phase: 2
  }
  phases {
    phase: 3
  }
}
pv_inverter {
  param {
  }
  phases {
    phase: 1
  }
  phases {
    phase: 2
  }
  phases {
    phase: 3
  }
}
]

hangingloose, 2025-02-27:
It fits also with my HYS

My problem is only, that I don´t know how to scroll in the Home Assistant Terminal... The last information was;

    phase: 1
    voltage: 2358
    current: 796
    active_power: 1378
    reactive_power: 1039
    power_factor: 80
    energy_frequency: 340
    energy_consumed: 2
  }
}
load {
  param {
    status: 1
    frequency: 5004
  }
  phases {
    phase: 1
    voltage: 2360
    active_power: -1352
    energy_consumed: 36
  }
}
inverter {
  param {
    status: 3
    frequency: 5002
    isolation_resistance: 5000
    leakage_current: 2
    drm_signal: 1
  }
  phases {
    phase: 1
    voltage: 2359
    current: 152
    active_power: -21
    reactive_power: 547
    dc_current: -8
    dc_voltage: 457
    eps_voltage: 2360
    eps_current: 26
    eps_power: -7
  }
}
generator {
  param {
  }
  phases {
    phase: 1
  }
}
pv_inverter {
  param {
  }
  phases {
    phase: 1
  }
}
power_flow {
  pv_to_grid: 1330
}
]

suaveolent, 2025-02-27:
@magyarl89 @hangingloose Well isn't that something we could work with :) thank you for testing!

Now that the hard work is done the next step will be to integrate it into home assistant.

I will take me a while though, since I need to rewrite part of the integration to handle the new requests and to differ between the old ad new inverters. 

Stay tuned. I will let you know once there is something you could test!

hangingloose, 2025-02-27:
I added my result above.
Should be complete now ;-)

But I have one additional wish...
The hybrid inverter has different operating modes.
Ideally we could switch between modes and change limits if necessary.
I would like not to put any additional strain on our power grid, so I would like to charge the battery at the time with the highest yield.

As an example that, I might not fully charge it until midday in the summer. Or charge the battery from the grid at certain times with dynamic electricity tariffs. 
To do this, I would need access to the values/switches via Home assistant

suaveolent, 2025-02-27:
@hangingloose could you maybe capture the requests the same way @magyarl89 ? It really helps me a lot. Since I then know what to look for.

hangingloose, 2025-02-27:
??? what kind of requests?
I posted the results of hoymiles-wifi --host 192.168.2.101 get-energy-storage-data

[113 more replies]

closed 2025-08-05 by suaveolent