suaveolent/ha-hoymiles-wifi #93: Missing model

BuchAir, 2025-01-30:
Detected that custom integration 'hoymiles_wifi' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('hoymiles_wifi', 'xxx'), with device info: {'identifiers': {('hoymiles_wifi', 'xxx')}, 'manufacturer': 'Hoymiles', 'model': 'HMS-1600/1800/2000-4T', 'name': 'Wechselrichter', 'serial_number': 'xxx', 'via_device': ('hoymiles_wifi', 'xxx')} at custom_components/hoymiles_wifi/sensor.py, line 674: async_add_entities(sensors). This will stop working in Home Assistant 2025.12.0

Would you be able to update?

suaveolent, 2025-01-30:
I think this is a relict from earlier versions of the integration. Can you try to "Reconfigure" the integration? Just click on the three vertical dots next to your integration entry and click on the "Reconfigure" button.

Toutatis64, 2025-02-07:
I also add an issue after the update of latest version, I lost all the values. A reconfigure solved my issues, thx !

closed 2025-03-24 by github-actions[bot]