suaveolent/ha-hoymiles-wifi #94: Restart Inverter

la-sina, 2025-02-19:
I’m experiencing an issue with my HMS-800-2T inverter when trying to output power values below 150 watts from my battery. The only way to resolve the issue is by restarting the inverter using the app.

Would it be possible to add a “Restart” button alongside the existing Power On and Power Off buttons? This would make it much more convenient to resolve using an automation when the error occurs

suaveolent, 2025-02-19:
Have you already tried the "Restart" button which you can find in the DTU? This should be the same button as you can find in the app.

suaveolent, 2025-02-19:
> I tried using the dtu restart button but I think it only restarts the dtu and not the inverter? In the app I can also see two separate restart buttons one for the dtu and one for the inverter so I assumed that they do different things? I‘m not sure. I also tried using the off and on button with a delay in between in my automation, but this did not help with my issue.

Thanks for confirming, I need to check wether this is a new functionality. I was no aware that there now is also a restart button for the inverter.

suaveolent, 2025-02-19:
> Just to clarify those are the two buttons I mean
> 
> [image]

Thanks, I must admit I have used the app for quite a while, so I will check.

N3rdix, 2025-03-04:
Same for me, the restart does not work as expected (e.g. After a power outage at day).
I need to restart via the S-Miles App to make it producing energy again

rahe71, 2025-05-28:
I appreciate an inverter restart button as well. I have to unplug and plug the inverter from time to time to change it from one socket to another. If I plug it less than approximately 30 seconds after I unplugged it, the inverter stops working until I restart it - switching off and on doesn’t work. If I plug it after one or more minutes after unplugging, I don’t need to restart. However, I would like to plug it immediately after unplugging it, so the inverter restart button would help a lot. Thanks in advance.

suaveolent, 2025-07-02:
@rahe71 I just merged the Pull-Request here https://github.com/suaveolent/ha-hoymiles-wifi/pull/95.

I still need to publish a new version though. In the meantime, you can update manually to the `main` branch.

closed 2025-08-05 by suaveolent