suaveolent/ha-hoymiles-wifi #96: daily energy reset not any more at midnight

deluxestyle, 2025-02-25:
**Describe the bug**
Since two days the sensor.hoymiles_hms_800w_t2_ac_daily_energy is not reseted anymore at midnight but only when the dtu is coming back online
This is a problem as I use it to claculate my total engery consumption

**To Reproduce**

[image]

**Expected behavior**
the daily energy should be set to  0 at midnight

**Screenshots**
see above

**Desktop (please complete the following information):**
N/A

**Smartphone (please complete the following information):**
N/A

deluxestyle, 2025-02-25:
I'm using 0.3.5 of the integration and Home Assitant 2025.2.5
Restart happened, but has not changed anything
no change in the last days on my set up.

deluxestyle, 2025-02-26:
today it is working again. not sure what happend.

suaveolent, 2025-02-26:
Thanks for letting me know. Maybe there was an issue with how the timers were propagaded via Home Assistant. Closing it for now.

deluxestyle, 2025-02-26:
I guess something like this. have seen that the whole system used a lot of ram during the last three days. did now a full restart of the whole system. maybe this is related

deluxestyle, 2025-03-15:
I would like to open it again @suaveolent 
it happened again this night

see screenshots

it happened for the ac daily energy

[image]

but not for the port_1_dc_daily_energy or port_2_dc_daily_energy

[image]

[image]

suaveolent, 2025-06-29:
It’s a really interesting phenomenon. It seems like the reset is called too late. Although I don’t understand why yet. Might be something specific to home assistant.

deluxestyle, 2025-06-29:
Any information that I can provide to help to solve this issue

suaveolent, 2025-06-30:
I will add some more debug statements to the next version. Then we can hopefully see what is going on.

PeterZechel, 2025-07-16:
I have also a comparable problem with the value. 
it freezes every night around 00:58 and needs a reset of the the service in HA to work again. All other values are provided like expected. 

[image]

closed 2026-10-13 by suaveolent