suaveolent/ha-hoymiles-wifi #98: Provide inverter state

saxn-paule, 2025-02-28:
I use the on/off buttons of the micro inverter to control the charging/discharging of my solar battery.

Is it possible to provide an entity that shows the current inverter state (on or off)?

In the S-Miles cloud it isn't visible weather it is on or off. But in off state an alarm is shown.

[image]

[image]

suaveolent, 2025-02-28:
It might be able to get this information by checking the errors/warnings. 
Unfortunately neither is this directly provided by the curent data nor have I yet found the necessary information in the app.

I'll add it as a Feature Request for future versions.

N3rdix, 2025-03-04:
Couldn't be the entities used representing the error code of each DC string?

[image]

saxn-paule, 2025-03-04:
In my case there is no error code when I turn the inverter off.