suaveolent/ha-hoymiles-wifi #99: No sync of power limit HA <-> Cloud and resetting its value to 0% at night by itself

clouskey, 2025-03-05:
**Describe the bug**
Active Output Power doesn't synchronize with Power limit set in Home Assistant. Active Output Power set in S-Miles Cloud doesnt' change Power limit value in HA. And vice versa - power limit set on HA doesn't set value in S-Miles Cloud.

Also power limit resets to 0% everyday at about 01:20 in the night in HA. Value in S-Miles Cloud is untouched but 0% affects real energy production despite of value set on S-Miles Cloud.

I have set Active Output Power in Cloud to 100% and I only change Power Limit in HA. It seems like DTU doesn't sync data to cloud.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Plant O&M on S-Miles Cloud.
2. Click on Power Control.
3. Set Active Output Power in % to e.g. 100%.
4. Go to HA and set another power limit.
5. Values don't synchronize at all.
6. At about 1:20 AM on HA power limit is set to 0% by itself every night (had to make automation to set to my preffered value).

**Additional context**
I've noticed that last change of power limit whether on S-miles cloud or in HA is taken into effect, even if you change Active Output Power in S-Miles Cloud and it doesn't change in HA it is set and working so you don't really know what current power limit is. Any power changes in HA aren't reported to S-Miles Cloud. I am using DTU-WLite-S with newest firmware (V00.04.00; before update there was the same bug; HW H06.02.01) and HMS-2000-4T (firmware V01.00.27; HW H00.04.10).

[image]

[image]

suaveolent, 2025-03-05:
Thanks for reporting and the detailed description. I'll have a look at it, although unfortunately I do not have a DTU-WLite-S.

clouskey, 2025-03-05:
> Thanks for reporting and the detailed description. I'll have a look at it, although unfortunately I do not have a DTU-WLite-S.

I suppose that there is a RAM in this DTU and it resets everyday and not push values to cloud, and after purge of RAM its not pulling data from cloud. Maybe you could debug this somehow. I turned on debug logging but I don't know where are the logs.

suaveolent, 2025-03-05:
For home assistant you should find your logs under "Settings" - "System" - "Logs", although I assume you won't find much information in there.

I am setting the power limit via the same functionality the app uses. The same goes true for reading the configuration.

What happens if you directly connect to your DTU via the Hoymiles App and set the power limit? Does it reset too or is that reflected in the cloud?

clouskey, 2025-03-05:
Do you mean connecting directly to WiFi / bluetooth of DTU?

suaveolent, 2025-03-05:
Yes exactly. In the app, there is a so called "toolkit", which should allow you to set the power limit too.

clouskey, 2025-03-05:
Thanks. I'll try that when I get home (in 5 hours). I will change Active Output Power if it is possible to and check Cloud after a while. Will be back with info.

clouskey, 2025-03-05:
Despite that there is still problem with resetting power limit value in HA

clouskey, 2025-03-05:
I don't see any option according to power limiting in local toolset. I can't change almost any options from there, just stats. I can change power limit only in cloud mode. 

[image]
[image]
[image]

suaveolent, 2025-03-05:
I see. 
That could explain why the setting is not synchronized to the cloud, since it was never intended to be synchronized back. 

What you can do is try blocking Internet access in the night, and see whether the DTU still resets the power limit.

clouskey, 2025-03-06:
I have a little trouble with banning internet connection to DTU but I did some logs and debugging by myself.

In my AP constroller there is log entry that says that DTU is reconnecting. I checked another devices and only this reconnects at night:
[image]

This is exact timestamp of power limit change:
[image]

This is the log from HA from this time range:
[home-assistant_hoymiles_wifi_2025-03-06T08-22-16.406Z.log](https://github.com/user-attachments/files/19105067/home-assistant_hoymiles_wifi_2025-03-06T08-22-16.406Z.log)

clouskey, 2025-03-07:
I did another tests last time. In the evening I turned off the integration from HA and set the power limit to 4% using cloud. In the night in my AP software there is still DTU reconnection. In the morning I turned on the integration and... power is set to 0 BUT it works correctly (4% set in cloud yesterday) - power is limited to desired value.
After all I suppose that 0% after night is bugged value only shown in HA after reboot of DTU but DTU uses value set in cloud instead.
There is still something wrong with syncing this value to cloud as I said earlier.

GLE-HomeAssistant, 2025-05-06:
Hi there!
I noticed the same behavior:
Every time, when the DTU restarts or the connection was lost the value of the DTU power limit is zero.
If the power limit was changed e.g. in the hoymiles app or hoymiles website, the value will change to. So I think the
value only refreshes with a value change.
If I change the value in the app or on website, it takes aproximately 5 minutes until the value in HA refreshes.
If I change the value in HA then only the value in HA changeds There is no update in the app or website.
My equipment: 
Hoymiles HMS 1800-4T
Hoymiles DTU-WLite-S
I hope this helps fixing the bug, because I want to control my Anker Solix discharge with the DTU power limit.
Thank you in advance!
GLE

fipiblitz, 2025-05-15:
Same here:
1) Power limit is zero in HA each morning; DTU us running 24/7 - I am not aware of nightly DTU restarts.
2) If I change the value in the app or on website, it takes a few minutes until the value in HA refreshes. This is OK
3) If I change the value in HA then only the value in HA changes. There is no update to the app or website.

suaveolent, 2026-10-27:
#229 improved power limit readback by using live inverter data when available. 

I’m closing this issue given its age and the recent changes. If the incorrect Home Assistant reading persists with the latest release containing #229, please reopen again.

closed 2026-10-27 by suaveolent