suaveolent/ha-hoymiles-wifi #100: "500 Internal Server Error Server got itself in trouble" on Vanilla Rpi5 installation

eximius313, 2025-03-09:
**Describe the bug**
I encounter "500 Internal Server Error Server got itself in trouble" but I'm not using Docker. 

**To Reproduce**
Steps to reproduce the behavior:
1. Vanilla HomeAssistant installation using Rpi5 Imager (newest version)
2. Install HACS
3. Add this repository
4. Download ha-hoymiles-wifi 
5. Go to "Settings" - "Devices & Services"
6. Click "Add integration" and Select "Hoymiles"
7. Error appears

**Expected behavior**
Error does not appear

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Rpi5, newest version
 - Browser: Chrome

suaveolent, 2025-03-10:
Can you look into the logs to see what kind of error is reported?

suaveolent, 2025-03-10:
@eximius313 Can you try again? It seems that it could not resolve the URL:

https://files.pythonhosted.org/packages/81/a6/a55557e1c034f8d0ffc894ca5ab820a3f5dfe0d89d9af1610b24a7094f9f/hoymiles_wifi-0.3.4-py3-none-any.whl.metadata

Which works for me. Maybe there was a short outage?

eximius313, 2025-03-10:
I've tried - it's the same

suaveolent, 2025-03-10:
@eximius313 can you try the following command from your RPI: 

`curl https://files.pythonhosted.org/packages/81/a6/a55557e1c034f8d0ffc894ca5ab820a3f5dfe0d89d9af1610b24a7094f9f/hoymiles_wifi-0.3.4-py3-none-any.whl.metadata`

eximius313, 2025-03-10:
Sure!
As I'm using HomeAssistant OS @ Rpi, I've used this plugin: https://www.home-assistant.io/integrations/shell_command/ in order to execute the command and here is the output:

```
stdout: >-
  Metadata-Version: 2.2

  Name: hoymiles-wifi

  Version: 0.3.4

  Summary: This Python library facilitates communication with Hoymiles HMS
  microinverters, specifically targeting the HMS-XXXXW-T2 series.

  Author: suaveolent

  Project-URL: Homepage, https://github.com/suaveolent/hoymiles-wifi

  Project-URL: Issues, https://github.com/suaveolent/hoymiles-wifi/issues

  Classifier: Programming Language :: Python :: 3

  Classifier: License :: OSI Approved :: MIT License

  Classifier: Operating System :: OS Independent

  Requires-Python: >=3.7

  Description-Content-Type: text/markdown

  License-File: LICENSE

  Requires-Dist: protobuf

  Requires-Dist: crcmod

  Dynamic: requires-dist

  # hoymiles-wifi

  This Python library facilitates communication with Hoymiles DTUs and the
  HMS-XXXXW-2T HMS microinverters, utilizing protobuf messages.

  For the Home Assistant integration have a look here:

  https://github.com/suaveolent/ha-hoymiles-wifi

  > [!NOTE]

  > Disclaimer: This library is not affiliated with Hoymiles. It is an
  independent project developed to provide tools for interacting with Hoymiles
  DTUs and Hoymiles HMS-XXXXW series micro-inverters featuring integrated WiFi
  DTU. Any trademarks or product names mentioned are the property of their
  respective owners.

  ## Supported Devices

  The library was successfully tested with:

  - Hoymiles HMS-400W-1T

  - Hoymiles HMS-800W-2T

  - Hoymiles HMS-2000DW-4T

  - Hoymiles DTU-WLite

  - Hoymiles DTU-Pro (S)

  - Solenso  H-1000 (not tested for command, only to get data)
[... 159 lines cut ...]
  > [!CAUTION]

  > Setting the update interval to 1–2 seconds could also cause disruptions in
  the App's connection.

  ### Compatibility:

  > [!IMPORTANT]

  > While developed for the HMS-800W-2T inverter, compatibility with other
  inverters from the series is untested at the time of writing. Exercise caution
  and conduct thorough testing if using with different inverter models.

  ## Attribution

  A special thank you for the inspiration and codebase to:

  - [DennisOSRM](https://github.com/DennisOSRM):
  [hms-mqtt-publisher](https://github.com/DennisOSRM/hms-mqtt-publisher)

  - [henkwiedig](https://github.com/henkwiedig):
  [Hoymiles-DTU-Proto](https://github.com/henkwiedig/Hoymiles-DTU-Proto)
stderr: "% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current\n                                 Dload  Upload   Total   Spent    Left  Speed\n\r  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0\r100  5853  100  5853    0     0  16134      0 --:--:-- --:--:-- --:--:-- 16168"
returncode: 0
```

suaveolent, 2025-03-11:
@eximius313 Good. Now let's try to tackle the remaining issue :) 

What kind of device are you using ? Are you sure the IP-address is correct?

eximius313, 2025-03-11:
1. DTU-Pro (WiFi)
2. Yes, I can ping the device
3. Shall I `nmap` it?

suaveolent, 2025-03-11:
Is your DTU connected via Ethernet? There have been some reports that it only works when connected to WiFi.

Updating to the newest firmware of the DTU might also enable the Ethernet port.

eximius313, 2025-03-11:
1. It's connected via WiFi
2. What firmware version are you using?
3. What features must be enabled (I was in contact with the manufacturer, so I could ask their support

suaveolent, 2025-03-11:
I don’t have a DTU Pro unfortunately I am running a HMS with integrated DTU. 

Can you check whether port 10081 is open on the DTU?

eximius313, 2025-03-11:
Yes, it's open

closed 2025-03-11 by eximius313

suaveolent, 2025-03-11:
> Ok, after another reinstall it finally worked! I don't know why there were so many connection issues, but I'm glad that it's finally working!

Great, thanks for letting me know :)

eximius313, 2025-03-11:
I thank you.
By the way - maybe you could catch this `Failed to fetch` Exception and provide the user with meaningful error message along with troubleshooting hints?

suaveolent, 2025-03-12:
@eximius313  Unfortunately, the error you see is specific to HA/HACS and cannot be caught in the integration, since it is trying to install it.