suaveolent/ha-hoymiles-wifi #101: Erreur Le flux de configuration n'a pas pu être chargé: {"message":"Invalid handler specified"}

arthur2017-ui, 2025-03-13:
I cannot get the integration from ADD integration button, after I had downloaded the repository : I have seen all the files in the folder= /homeassistant/custom_components/hoymiles_wifi/ : i get the message= Erreur
Le flux de configuration n'a pas pu être chargé: {"message":"Invalid handler specified"} .... what shoud I check and do ?

arthur2017-ui, 2025-03-14:
I cannot get the integration from ADD integration button, after I had downloaded the repository : I have seen all the files in the folder= `/homeassistant/custom_components/hoymiles_wifi/ ` 
I get the error message= ` Erreur
Le flux de configuration n'a pas pu être chargé: {"message":"Invalid handler specified"}`

Could you please let me know what I have to check and do ?

suaveolent, 2025-03-14:
Can you have a look at the log files and paste the full error here?

suaveolent, 2025-03-14:
What home assistant version are you on?

arthur2017-ui, 2025-03-14:
`Home Assistant

Core
2024.2.1
Supervisor
2025.03.2
Operating System
11.4
Interface utilisateur
20240207.1`

arthur2017-ui, 2025-03-14:
`Raspberry Pi 3

rpi3-64`

suaveolent, 2025-03-14:
Got it thanks, you need to update your home assistant core installation.

arthur2017-ui, 2025-03-14:
Vous voulez dire le Core ?

suaveolent, 2025-03-14:
The current home assistant version is 2025.3.3.

arthur2017-ui, 2025-03-14:
Comme j'ai seulement un RPI 3B+, j'ai peur que les ressources ne vont pas suffire à faire tourner la nouvelle version du core ? .... si la nouvelle version demande beaucoup plus de ressources ?

suaveolent, 2025-03-14:
> Comme j'ai seulement un RPI 3B+, j'ai peur que les ressources ne vont pas suffire à faire tourner la nouvelle version du core ? .... si la nouvelle version demande beaucoup plus de ressources ?

It should also work on a Raspberry pi 3B. Although I cannot say for the supervisor version. 

I would make a backup and try updating to see if there are any problems .

arthur2017-ui, 2025-03-14:
Je viens d'installer la version 2025.3.3.
Et j'ai bien mon DTU et micro-onduleur qui s'affichent !
Merci beaucoup et BRAVO pour le développement

closed 2025-03-14 by arthur2017-ui

suaveolent, 2025-03-14:
You're more than welcome!