suaveolent/ha-hoymiles-wifi #111: dtu_ac_tagesenergie only 3Wh per day

rt1970, 2025-05-03:
Hi!

My sensor dtu_ac_tagesenergie shows only 1, 2 or 3Wh per day.
I think it is kWh, but show no decimal places.

How can i change this?

I have only this in config.yaml

homeassistant:
  packages: !include_dir_named packages

# Laube  
  customize:
    sensor.laube_strom_laube_verbrauch:
      device_class: energy
      unit_of_measurement: "kWh"
      state_class: total_increasing

    sensor.laube_strom_laube_einspeisung:
      device_class: energy
      unit_of_measurement: "kWh"
      state_class: total_increasing

    sensor.dtu_ac_tagesenergie:
      device_class: energy
      unit_of_measurement: "kWh"
      state_class: daily_increasing

Steffen06, 2025-05-07:
I am not the owner but I have a current knowledge about the Project and can try to help you. The sensor unit of Wh should be the correct one and I think that you have some problem with the communication with the DTU. Can you provide more Information about your problem:

- What settings do you use:

rt1970, 2025-05-20:
I have Hoymiles HMS-800W/1000W-2T.
DTU is integrated: DTUBI from Hoymiles
HA-Plugin: Version 0.3.5
sensor.dtu_ac_tagesenergie show only full kWh without decimal.
In S-Miles App does show 4.83 kWh but HA show 5kWh.
But the Graph shows many Points but without decimal...

[image]

Steffen06, 2025-05-20:
@rt1970 Ok that Information helps. I have an identical set up, but if I Compare it to mine and the configuration the than things start seems a little odd.

- [x] If compare the graph with you value the date seems to be correct and received with full resolution
- [ ] According to the configuration the Value should be in Wh and not in kWh, but is scaled correctly
- [ ] The Value gets rounded in the frontend, otherwise you would see only 5 steps and not a steadily increasing curve

My guess would be that you have somehow overwritten the unit_of_measurement (and rounding of the displayed value) in a HA configuration. Could you check when you hover over the graph if you get the full resolution (0,001 kWh / 1 Wh steps).  And check if you have configured any thing to overwrite the unit / rounding.

[image]

https://github.com/suaveolent/ha-hoymiles-wifi/blob/8e0a26c4dfa70cd95fddf3558938fad786572e37/custom_components/hoymiles_wifi/sensor.py#L92-L111

rt1970, 2025-05-21:
sensor.py is original

The only is in config.yaml:
# Laube  
  customize:
    sensor.laube_strom_laube_verbrauch:
      device_class: energy
      unit_of_measurement: "kWh"
      state_class: total_increasing

    sensor.laube_strom_laube_einspeisung:
      device_class: energy
      unit_of_measurement: "kWh"
      state_class: total_increasing

    sensor.dtu_ac_tagesenergie:
      device_class: energy
      unit_of_measurement: "kWh"
      state_class: daily_increasing

I have export and Excel show this and many other Wh:
sensor.dtu_ac_tagesenergie,2.057,2025-05-21T08:56:54.623Z
sensor.dtu_ac_tagesenergie,2.068,2025-05-21T08:57:52.555Z

I have found my fault!!! 🤝
Wrong definition 😤
Here is now my correct config:

[image]

closed 2025-06-21 by rt1970