suaveolent/ha-hoymiles-wifi #114: No data from DTU?

stefanoneri, 2025-05-12:
Hi there! 

I am currently having the following issue: I am using an Hoymiles 800w2t connected through wifi with my home network. Homeassistant does not seem to have received any data since yesterday around 4pm German time. When I check with the S-Miles Installer App, I can see, that the unit is producing and feeding power to my grid (Img1) vs (Img2 - view in Homeassistant). Neither restarting the DTU, Homeassistant Service nor the System itself helped so far. Any other ideas are greatly appreciated! 

[image]

vs.:

[image]

Steffen06, 2025-05-12:
Have you checked the following things:
- Is the DTU connected to your WiFi?
- Can you ping the DTU?
- Is the configured IP still correct (assigned by your DHCP server) / DNS entry working?

stefanoneri, 2025-05-13:
Yes, as I mentioned - with the S-Miles Installer App, I can still read data from the DTU (being in my normal home Wifi). These are the reading from the App today.

- Network config did not change (<- I was soo wrong here)
- I can ping the DTU

After working through your list I compared the IP address which was configured in the Integration with the real one given by the router and found out that they deviate... 

Working again! Thank you!