suaveolent/ha-hoymiles-wifi #115: Hoymiles integration broken after updating to HA 2025.5.2

SebboGit, 2025-05-18:
After updating to 2025.5.2 my Hoymiles DTU and Hoymiles Inverter devices were gone.
Checking the logs I get this error:

`Logger: homeassistant.setup
Source: setup.py:344
First occurred: 2:57:28 PM (1 occurrence)
Last logged: 2:57:28 PM

Setup failed for custom integration 'hoymiles_wifi': Unable to import component: No module named 'hoymiles_wifi'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 344, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1013, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 993, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1053, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/hoymiles_wifi/__init__.py", line 12, in <module>
    from hoymiles_wifi.dtu import DTU
ModuleNotFoundError: No module named 'hoymiles_wifi'`

sid-dich, 2025-05-21:
Hi i am not sure if my problem relates to this issue. I am not able to add the repository to my HA. I get the error

[image]

BR
Sid-Dich

MarioWDS, 2025-05-21:
Ich habe zuerst das Fehlen der bisher stabil kommenden Inverterdaten bemerkt und bin bei der Suche auch hierauf aufmerksam geworden:

[image]

Mein HA Update war auch auf 2025.5.2.

SebboGit, 2025-05-21:
> Are you sure the required python module is installed? I've just upgraded my dockerized HA to 2025.5.2 and set-up the hoymiles integration without any hiccups
 
Pretty sure the module is installed since it was working just fine in 2025.5.1 and I did nothing but update to 2025.5.2.

SebboGit, 2025-05-21:
> Ich habe zuerst das Fehlen der bisher stabil kommenden Inverterdaten bemerkt und bin bei der Suche auch hierauf aufmerksam geworden:
> 
> [image]
> 
> Mein HA Update war auch auf 2025.5.2.

Yep, same exact error I am getting.

SebboGit, 2025-05-22:
> Thanks for reporting. Unfortunately I am currently swamped with work so I cannot have a look until maybe next weekend. Can you have a look in the log if there is anything in there?

Thanks for looking into it. I know this is not something you'd wana hear as a developer but unfortunately I did restore a backup right after the update broke the integration so I can't provide any more logs. However, I did give updating a second try and I am now on 2025.5.2. and the integration is working. I'd mark this as closed now since you obviously can upgrade without it breaking. Don't know what the issue was the first time.

> How is your HA installed? Could you please attach the output of "pip freeze" from the terminal?

I am running HA as a VM on a Proxmox host.

closed 2025-05-22 by SebboGit

MarioWDS, 2025-05-24:
Todays update seems to fix the problem.