suaveolent/ha-hoymiles-wifi #125: Failed to connect

salfuman33, 2025-07-25:
I have a HYS-6.0LV-EUG1 Hoymiles inverter.

I followed the steps in the documentation but it doesnt works fine.

I can get some info, but other doesnt works:

[image]

I can change the working mode with the CLI into interactive mode.

And when I tried to use the integration with home assistant, I get an error all the time:

[image]

Maybe im doing something wrong, but I am lost with this.

Many thanks.

salfuman33, 2025-07-25:
May be I need to install pip install hoymiles-wifi into the virtual environment and not into the global terminal?
Im trying to do this, but I dont know how to do it:
https://github.com/suaveolent/hoymiles-wifi/issues/68#issuecomment-3050141309

suaveolent, 2025-07-25:
@salfuman33 see here how to do that: https://github.com/suaveolent/ha-hoymiles-wifi/issues/88#issuecomment-3034933498

salfuman33, 2025-07-25:
> [@salfuman33](https://github.com/salfuman33) see here how to do that: [#88 (comment)](https://github.com/suaveolent/ha-hoymiles-wifi/issues/88#issuecomment-3034933498)

Now its working! Sorry I dont see it before. I have installed the "energy" repo and now works:

[image]

I see that no actions are available. Its on roadmap the hability to change inverter working mode? I could change it using the CLI, but could be more interesting do it through HA directly

suaveolent, 2025-07-26:
@salfuman33 yes, it’s planned but it will take a while to implement it. Especially since most working modes require some parameters it will o my be possible by service calls in home assistant.

salfuman33, 2025-07-27:
Now all its working.
The change of the working modes will be done by external scripts until the integration includes it.

Many many thanks!

closed 2025-07-27 by salfuman33