suaveolent/ha-hoymiles-wifi #133: Dependency unresolved after upgrade to 0.4.0

gtessa, 2025-08-06:
**Describe the bug**
After allowing the automatic update, the following error log appears, and the integration is disabled.

```
2025-08-06 14:56:51.602 ERROR (SyncWorker_4) [homeassistant.util.package] Unable to install package hoymiles-wifi==0.5.0: × No solution found when resolving dependencies:
  ╰─> Because hoymiles-wifi==0.5.0 depends on protobuf>=5.29.3 and
      protobuf==5.29.2, we can conclude that hoymiles-wifi==0.5.0 cannot be
      used.
      And because you require hoymiles-wifi==0.5.0, we can conclude that your
      requirements are unsatisfiable.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Using previous version, select update in Home Assistant notification .

**Expected behavior**
No error

**Screenshots**
N/A

**Additional context**
Running HA as docker.

suaveolent, 2025-08-06:
@gtessa which home assistant version have you installed?

suaveolent, 2025-08-06:
@gtessa Thanks. Yes, there seems to be an issue on my side given the minimum Home Assistant version.

Also, since you mentioned you are running Core. With 2025.6 Core is deprecated and you need to switch to a container based system or Home Assistant OS.

See:

https://www.home-assistant.io/blog/2025/05/22/deprecating-core-and-supervised-installation-methods-and-32-bit-systems

closed 2025-09-21 by github-actions[bot]