suaveolent/ha-hoymiles-wifi #134: S-Miles app and Home Assistant conflict

Jorn1993, 2025-08-07:
**Describe the bug**
I still am having issues using s-miles and the hoymiles integration together. When activiting the hoymiles integration, it looks like the s-miles app is not updating anymore. Even when i increase the timing to more than 32 seconds. There is just a gap in the s-miles data. 

V0.4.0 

[image]

[image]

[image]

**Expected behavior**
s-miles app and home assistant to work together.

fog191, 2025-08-07:
Many thanks to Jorn for opening this issue, as well as the answers given. I encountered exactly the same behaviour after updating the DTU firmware from 00.06.01 to 01.01.00. After some hours/days of playing around with the Hoymiles App and the HA integration I came to the same conclusion that concurrent access from both sides to the datastream is not longer possible. I have send an e-mail to the Hoymiles support team asking for a fix and they finally downgraded my DTU back to 06.01. Now it works again, at least as long the upgrade to the lastest firmware is not enforced.

Jorn1993, 2025-08-07:
[image]

[image]

With 90 seconds, for now, I can see the data in both s-miles and HA

MarkusRiLi, 2025-08-09:
I’m observing the following behavior: When the update interval is newly set, it works for a few hours and then stops again. It seems almost irrelevant which value is chosen. I had recently set it to 1000 seconds. It worked for a few hours, and then the connection to HomeAssistant broke. However, the cloud continued to receive data.

suaveolent, 2025-08-09:
@MarkusRiLi thanks for reporting. It might very well be that the random encryption data is changing after a while. Currently I am setting it only once. I can push an update to see if it has changed. Maybe that fixes the issue.

Memelhemel, 2025-08-11:
I have the same problems. @suaveolent if you push a fix I am happy to test it.

suaveolent, 2025-08-11:
Let's track the issue here:

https://github.com/suaveolent/ha-hoymiles-wifi/issues/138

closed 2025-08-11 by suaveolent