suaveolent/ha-hoymiles-wifi #136: Connectivity broken with WLite-S running V01.01.00

elmodem, 2025-08-09:
**Describe the bug**
The integration is running smoothly with version 00.03.13.
When I updated one of my two Wlite-S sticks to V01.01.00, this stick only appears as "disconnected."
The stick's status in the Hoymiles app is OK, and everything works there.
Removing and re-adding the stick with version V01.01.00 also only provided a brief solution. The stick remained connected for about two hours, but didn't display any further information other than the connection status and signal strength. But during this time, values came back from the inverter.
Since then, the stick has been offline again.

**To Reproduce**
idk. V01.01.00?

**Expected behavior**
The stick is connected and outputs the values of the inverter. 

**Screenshots**
none

**Versions**
HA OS: 16.0
Core: 2025.8.0
Hoymiles integrtion: 0.4.0

**Related output from bug log**
2025-08-09 08:51:45.437 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hoymiles_wifi' calls `device_registry.async_get_or_create` referencing a non existing `via_device` ('hoymiles_wifi', '***removed***'), with device info: {'identifiers': {('hoymiles_wifi', '***removed***')}, 'manufacturer': 'Hoymiles', 'model': 'HMS-1600/1800/2000-4T', 'name': 'Wechselrichter', 'serial_number': '***removed***', 'via_device': ('hoymiles_wifi', '***removed***')} at custom_components/hoymiles_wifi/sensor.py, line 1209: async_add_entities(sensors). This will stop working in Home Assistant 2025.12.0, please create a bug report at https://github.com/suaveolent/ha-hoymiles-wifi/issues

2025-08-09 09:29:45.462 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Setup entry with update interval 0:01:00. IP: 192.168.220.57
2025-08-09 09:29:45.472 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Setup entry with update interval 0:05:00. IP: 192.168.220.57
2025-08-09 09:29:45.478 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Setup entry with update interval 2:00:00. IP: 192.168.220.57

2025-08-09 09:29:50.515 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Unable to retrieve real data new. Inverter might be offline.
2025-08-09 09:29:50.515 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Finished fetching hoymiles_wifi data in 5.009 seconds (success: True)

2025-08-09 09:29:50.517 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Hoymiles data coordinator update
2025-08-09 09:29:50.522 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Unable to retrieve config data. Inverter might be offline.
2025-08-09 09:29:50.522 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Finished fetching hoymiles_wifi data in 0.006 seconds (success: True)
2025-08-09 09:29:50.523 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Hoymiles data coordinator update
2025-08-09 09:29:50.530 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Unable to retrieve app information data. Inverter might be offline.
2025-08-09 09:29:50.530 DEBUG (MainThread) [custom_components.hoymiles_wifi.coordinator] Finished fetching hoymiles_wifi data in 0.007 seconds (success: True)

nordpol18, 2025-08-09:
Exactly the same issue: after Updating DTU Firnware from V00.04.00 to V01.01.00, integration to hone assistant fails. In contrast to @elmodem experience I was not asked to set a new DTU Access Point PW in the S-Miles app. 
As I did so manually, integration comes back again. (Also from DTU) Only the Control elements like power limitation seem not to work

[image]

tupacamaru7, 2025-08-09:
I can confirm that the power limitation is NOT working with the new firmware. I have deleted and added the entry (DTU and inverter), new name was provided to the power limitation, but it stays "unknown" ("Unbekannt") in Home Assistant, so that my automations regarding the power limitations are not working. That is very unfortunate ...

tupacamaru7, 2025-08-09:
This is quite strange. In my Home Assistant value is unknown (see attached screenshot) and can't be set with the slider. My automations are not able to change it. I can only change it manually in the Hoymiles app. Can you pls attach some screenshot how it looks in your Home Assistant and how you can slide/control it? I did several restarts/ reloads of Hoymiles/Home Assistant. Thanksnin advance!

[image]

nordpol18, 2025-08-09:
It is also greyed out on the Dashboard: [image]
But when accessing through the integrations page - Hoymiles, I can see still a slider that is functional (despite not reading value 'unknown') 

[image]

But when I slide it, it reduces Power:

[image]

tupacamaru7, 2025-08-09:
Ok, thanks for quick feedback. I will try tommorow during the day when it is sunny ... I can slide this one currently, but when i exit this screen and come back, the slider is again in the middle. I will test tommorow during the day again and will write here.

suaveolent, 2025-08-11:
Let's track these issues here:

https://github.com/suaveolent/ha-hoymiles-wifi/issues/137
https://github.com/suaveolent/ha-hoymiles-wifi/issues/138