suaveolent/ha-hoymiles-wifi #138: HA looses connection to DTU-WLite-S

wuk986, 2025-08-10:
Hi!

A big thank you for doing this project and the quick fix, to get this integration working again after the firmware update.
So far all is working like a charm and unfortunately it looks like, that Home Assistant is loosing the connection to my DTU-WLite-S.

So far I have no idea if the connection is closed by the DTU-WLite-S or Home Assistant is loosing it. For now I have to reconfigure the integration once a day in the morning to get the connection up again.

Jorn1993, 2025-08-10:
Same for me

suaveolent, 2025-08-11:
@Memelhemel  @Jorn1993  @dvbcheck @wuk986 I just pushed a possible fix to the `[update_enc_rand](https://github.com/suaveolent/ha-hoymiles-wifi/tree/update_enc_rand)`  branch.

If you update your integration to that version you can test wether it works. It periodically checks whether the encryption data have changed and if so, update it accordingly.

Here is how to do that:
https://hacs.xyz/docs/use/entities/update/#install-action

for version, you just type `update_enc_rand`

Jorn1993, 2025-08-12:
Everything looks good ! Still working after last night in HA. Also the s-miles app working fine.

dvbcheck, 2025-08-12:
DTU-WLite-S Firmware V01.01.00. 
Latest version of HA (docker on Raspi Pi 5 4GB) and updated version 0.40.
No I don ´t need to reconfigure the Home Assistant integration every morning.

But I can ´t receive correct data while S-Miles Home on iPad (picture below)/

[image]

After deleting the hoymiles Devices in HA (or shutting down HA), the data in in S-Miles App seems to be correct.

[image]

HolgerFoe, 2025-08-12:
> [@dvbcheck](https://github.com/dvbcheck) you might need to increase your Update interval. Try using 90s or more.

I used 120sec this morning and this worked fine.

suaveolent, 2025-08-15:
@wuk986 great thanks for the feedback. I will prepare a new release in the next days. 

Need to fix some sensor issues and also support for controlling the hybrid inverters first.

DeCysos, 2025-08-21:
> [@Memelhemel](https://github.com/Memelhemel) [@Jorn1993](https://github.com/Jorn1993) [@dvbcheck](https://github.com/dvbcheck) [@wuk986](https://github.com/wuk986) I just pushed a possible fix to [...]

Thanks, that works fine

suaveolent, 2025-08-22:
@DeCysos Did you restart you instance after updating to `update_enc_rand`? 

Also after manually installing the development branch, Home assistant suggests there is a new update, which would effectifely installe the release version and revert the changes.

MarkusRiLi, 2025-08-25:
> [@DeCysos](https://github.com/DeCysos) Did you restart you instance after updating to `update_enc_rand`?
> 
> Also after manually installing the development branch, Home assistant suggests there is a new update, which would effectifely installe the release version and revert the changes.

I also tried to install the update for `update_enc_rand`, but I didn’t receive a prompt for a new update. However, I’m a beginner with Home Assistant, so I’m not sure if I did everything correctly.

soulfire89, 2025-08-28:
[image]
Same issue here with integration version 0.4.0. The disconnect of the DTU happens every night at 1:03.00 +-10sec
For now a reconfiguration every morining helps.
Today I've set the update intervall to 90s and now waiting for tomorrow 1:03...

soulfire89, 2025-08-29:
@suaveolent problem solved for me, no disconnect tonight

Alex-Sun-sumy, 2025-08-30:
Please tell me when we can expect a release version with a fix for lost connection with DTU?

Jodelpeter, 2025-08-30:
> Go to `Developer tools` - `Actions` - Select `Install update (update.install)`. For target select `Hoymiles update` For version type `main`.

It worked very well for me too. This morning was the first time I didn't have to reconfigure the integration for the handshake. Great work! Keep it up!

dvbcheck, 2025-09-06:
Even with DTU-WLITE-s Firmware V01.01.02 everything is fine

jschl2910, 2025-09-08:
Yes i did this:

Go to Developer tools - Actions - Select Install update (update.install). For target select Hoymiles update For version type main.

No improvement for me…

suaveolent, 2025-09-08:
@jschl2910 hm. I assume you also did a reboot afterwards?

dvbcheck, 2025-09-08:
Set update intervall to 90 sec, since than no reconfiguration necessary.

suaveolent, 2025-09-10:
@jschl2910 you could turn on debug logging for hoymiles-wifi and check if a message like `Updating enc_rand in config entry and DTU from<> to <>` appears. If not, it means that it does not detect the changed `enc_rand` for whatever reason and we need to dig deeper.

jschl2910, 2025-09-10:
Ok thanks, i implemented the code in my configuration.yaml, How Can i look into the log? Where do i find it? 

So the idea is:
1. integeate the Code
2. Go to Developer tools - Actions - Select Install update (update.install). For target select Hoymiles update For version type main.
3. restart HA
4. look into the log (need help were to find)

Right?

suaveolent, 2025-09-10:
@jschl2910 Yes, exactly. You can find the log under "Settings" - "System" - "Logs"

jschl2910, 2025-09-10:
There Are no Logs from hoymiles… 

I do find this update, that i di not install right?

[image]

suaveolent, 2025-09-21:
@TimWelter this is a different issue. Have a look at the logs to see if there is anything explaining the issue. My guess is that there is something wrong with your network or you have the wrong ip/hostname. Are you using a guest network maybe?

TimWelter, 2025-09-21:
@suaveolent it just shows `RealDataNew and GatewayInfo is None. Cannot connect to DTU or invalid response received!` like I mentioned in my earlier comment https://github.com/suaveolent/ha-hoymiles-wifi/issues/138#issuecomment-3281172334.

I'm not on a guest network. I am able to ping the DTU's IP from the HA container

TimWelter, 2025-09-21:
@suaveolent I'll try and see how it goes with wifi. The guy that installed it said the wifi connection can be unstable. I did already have Hoymiles support update it to the latest version. Thanks for the tip.

suaveolent, 2025-09-21:
@TimWelter just note that with WiFi you will need to enter a different ip address. 

Also you could try if the get-real-data-new command from the library works from outside of the container (e.g. on another computer in the network): 

https://github.com/suaveolent/hoymiles-wifi

suaveolent, 2025-09-21:
@TimWelter can you try connecting directly to the hotspot of the DTU and see if the Hoymiles app shows any live data (not to be confused with the cloud data).

If yes, you could try connecting your computer directly to the DTU and see if it is now able to successfully issue the `get-real-data-new` command or even better, try the `app-information-data` command.

TimWelter, 2025-09-21:
@suaveolent nope doesn't work. The app just gives: `Request failed. Please check network`.

suaveolent, 2025-09-24:
@Mailied this is a different problem.
I am using these values to detect the inverter model, and apparently I have not yet seen it.

 Can you maybe create a new issue at https://github.com/suaveolent/hoymiles-wifi to track the issue? Also please make sure to include your inverter model s.t. I can add it to the list of models.

Sonnenwerk, 2025-10-01:
Is this solved now with the 0.5 ?

suaveolent, 2025-10-31:
Should be fixed with `v0.5.0` now.

Sonnenwerk, 2025-10-31:
{"created_at":"2024-12-07T17:28:08.124965+00:00","data":{},"disabled_by":null,"discovery_keys":{},"domain":"min_max","entry_id":"01JEH2M65W7KDYZEQFJGZWH6N6","minor_version":1,"modified_at":"2025-07-12T12:57:59.199156+00:00","options":{"entity_ids":["sensor.wechselrichter_gartenhaus_port_1_dc_tagesenergie","sensor.wechselrichter_gartenhaus_port_2_dc_tagesenergie","sensor.wechselrichter_gartenhaus_port_3_dc_tagesenergie","sensor.wechselrichter_gartenhaus_port_4_dc_tagesenergie","sensor.wechselrichter_gartenhaus_port_5_dc_tagesenergie","sensor.wechselrichter_gartenhaus_port_6_dc_tagesenergie"],"name":"wow","round_digits":2.0,"type":"sum"},"pref_disable_new_entities":false,"pref_disable_polling":false,"source":"user","subentries":[],"title":"Hoymiles Gartenhaus Daily Energy (DC Sum Kombi)","unique_id":null,"version":1},
      {"created_at":"2025-07-12T13:21:15.423975+00:00","data":{},"disabled_by":null,"discovery_keys":{},"domain":"integration","entry_id":"01JZZCR3MZ776QXT6KHXETZ2CD","minor_version":2,"modified_at":"2025-08-19T15:51:17.601370+00:00","options":{"max_sub_interval":{"hours":24,"minutes":0,"seconds":0},"method":"left","name":"Hoymiles Gartenhaus (integral)","source":"sensor.hoymiles_gartenhaus_daily_energy_kombi","unit_prefix":"k","unit_time":"h"},"pref_disable_new_entities":false,"pref_disable_polling":false,"source":"user","subentries":[],"title":"Hoymiles Gartenhaus (integral)","unique_id":null,"version":1},
      {"created_at":"2025-07-12T13:21:15.423975+00:00","data":{},"disabled_by":null,"discovery_keys":{},"domain":"integration","entry_id":"01JZZCR3MZ776QXT6KHXETZ2CD","minor_version":2,"modified_at":"2025-08-19T15:51:17.601370+00:00","options":{"max_sub_interval":{"hours":24,"minutes":0,"seconds":0},"method":"left","name":"Hoymiles Gartenhaus (integral)","source":"sensor.hoymiles_gartenhaus_daily_energy_kombi","unit_prefix":"k","unit_time":"h"},"pref_disable_new_entities":false,"pref_disable_polling":false,"source":"user","subentries":[],"title":"Hoymiles Gartenhaus (integral)","unique_id":null,"version":1},

I found three entries for Gartenhaus. And this one for the DTU. I set is_encrypted to true and restarted

      {"created_at":"1970-01-01T00:00:00+00:00","data":{"dtu_serial_number":"xyz","enc_rand":"xyt","host":"192.168.178.78","hybrid_inverters":[],"inverters":[],"is_encrypted":true,"meters":[],"ports":[{"inverter_serial_number":"xx","port_number":1},{"inverter_serial_number":"xx","port_number":2},{"inverter_serial_number":"xx","port_number":3},{"inverter_serial_number":"138290602905","port_number":4},{"inverter_serial_number":"xx","port_number":5},{"inverter_serial_number":"138290602905","port_number":6}],"three_phase_inverters":["xx"],"update_interval":35},"disabled_by":null,"discovery_keys":{},"domain":"hoymiles_wifi","entry_id":"01J2Y43D7YHCT96B988XC7EP7H","minor_version":1,"modified_at":"2025-10-31T16:15:37.969200+00:00","options":{},"pref_disable_new_entities":false,"pref_disable_polling":false,"source":"user","subentries":[],"title":"192.168.178.78 Gartenhaus","unique_id":"xx","version":5},

But i see no change atm

suaveolent, 2025-10-31:
@sonnenwerk thanks, I see that there are noninverters configured in the entry. Can you try to „reconfigure“ the integration ?

Sonnenwerk, 2025-10-31:
@suaveolent After reconfiguring, i get those two startup errors. But something has changed. I see Version infos again. I will wait until "sun".
Thanks for having a look into it !

2025-10-31 19:39:12.435 ERROR (MainThread) [custom_components.hoymiles_wifi]   hass_data: {'dtu_serial_number': 'xx', 'enc_rand': 'xx', 'host': '192.168.178.78', 'hybrid_inverters': [], 'inverters': [], 'is_encrypted': True, 'meters': [], 'ports': [{'inverter_serial_number': '138290602905', 'port_number': 1}, {'inverter_serial_number': 'x', 'port_number': 2}, {'inverter_serial_number': '138290602905', 'port_number': 3}, {'inverter_serial_number': 'x', 'port_number': 4}, {'inverter_serial_number': '138290602905', 'port_number': 5}, {'inverter_serial_number': 'x', 'port_number': 6}], 'three_phase_inverters': ['138290602905'], 'update_interval': 35, 'dtu': <hoymiles_wifi.dtu.DTU object at 0x7f5ea4a6bbb0>, 'data_coordinator': <custom_components.hoymiles_wifi.coordinator.HoymilesRealDataUpdateCoordinator object at 0x7f5eaa16e490>, 'config_coordinator': <custom_components.hoymiles_wifi.coordinator.HoymilesConfigUpdateCoordinator object at 0x7f5eaa16e350>, 'app_info_coordinator': <custom_components.hoymiles_wifi.coordinator.HoymilesAppInfoUpdateCoordinator object at 0x7f5eaa16d310>}
2025-10-31 19:39:12.435 ERROR (MainThread) [custom_components.hoymiles_wifi]   config_entry_id: 01J2Y43D7YHCT96B988XC7EP7H